adafruit/TFTLCD-Library #37: Problem with 'textsize'

PCov3r, 2019-07-20:
Hello, 
first of all, thank you for making such a good library. 

I tried using the example tftbmp on my arduino mega to display bmp on my screen 
(the screen model : https://www.amazon.fr/Kuman-Tactile-Arduino-Module-Fonction/dp/B075CXXL1M/ref=sr_1_1?ie=UTF8&qid=1528549257&sr=8-1&keywords=tft+arduino+uno )

Here's the problem. When I try to upload the code, I get the following error : 

TFTLCD-Library-master\Adafruit_TFTLCD.cpp:115:3: error: 'textsize' was not declared in this scope

   textsize  = 1;

How may I solve it ?

JAICHANGPARK, 2019-07-23:
Same problem

darkpanda08, 2019-07-26:
Use Adafruit GFX library version 1.5.3 The newer versions of this library have broken everything.

JAICHANGPARK, 2019-07-27:
@darkpanda08  Thanks

closed 2019-07-31 by PCov3r

SamuelDave, 2019-08-21:
issue still persists with mine, please help