adafruit/TFTLCD-Library #43: Needs updating to latest GFX library

caternuson, 2020-01-23:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=22&t=161254
and maybe more things?

makermelissa, 2020-07-27:
I tried this out and it's not throwing errors for me, but it's also not displaying anything.

makermelissa, 2020-07-27:
It's also not detecting the LCD Driver Chip:
```
TFT LCD test
Using Adafruit 2.8" TFT Breakout Board Pinout
TFT size is 240x320
Unknown LCD driver chip: 0
If using the Adafruit 2.8" TFT Arduino shield, the line:
  #define USE_ADAFRUIT_SHIELD_PINOUT
should appear in the library header (Adafruit_TFT.h).
If using the breakout board, it should NOT be #defined!
Also if using the breakout, double-check that all wiring
matches the tutorial.
```

makermelissa, 2020-07-27:
Forcing the ID to 0x9341 got it working, so I need to concentrate on the readID() function.

PR #48 (2020-07-27): Fix warnings and chip detection
Fixes #43. Also this fixes warnings so several duplicate PRs that are out of date can be closed.
closed 2020-07-27 by makermelissa