adafruit/TFTLCD-Library #44: adafruit_tftlcd.cpp: 'textsize' was not declared in this scope

i-make-robots, 2020-02-07:
- Arduino board:  UNO R3
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.10
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
Set board & port.
Open tftbmp example sketch.
Compile.

Error log in arduino:

[me]\Documents\Arduino\libraries\Adafruit_TFTLCD_Library\Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::init()':

[me]\Documents\Arduino\libraries\Adafruit_TFTLCD_Library\Adafruit_TFTLCD.cpp:115:3: error: 'textsize' was not declared in this scope

   textsize = 1;

   ^~~~~~~~

[me]\Documents\Arduino\libraries\Adafruit_TFTLCD_Library\Adafruit_TFTLCD.cpp:115:3: note: suggested alternative: 'textsize_x'

   textsize = 1;

   ^~~~~~~~

   textsize_x

Multiple libraries were found for "SPI.h"
 Used: [me]\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\SPI
Multiple libraries were found for "Adafruit_GFX.h"
 Used: [me]\Documents\Arduino\libraries\Adafruit_GFX_Library
Multiple libraries were found for "Adafruit_TFTLCD.h"
 Used: [me]\Documents\Arduino\libraries\Adafruit_TFTLCD_Library
Multiple libraries were found for "SD.h"
 Used: C:\Program
exit status 1
Error compiling for board Arduino Uno.

------------------------------
The arduino error cuts off on the space in "Program Files"?!  Wow.

Thank you!  Stay awesome!

ladyada, 2020-02-07:
please try https://github.com/adafruit/TFTLCD-Library/pull/42

ladyada, 2020-02-07:
https://github.com/adafruit/TFTLCD-Library/pull/42 fixed

closed 2020-02-07 by ladyada

TuakDaunMerah, 2023-08-23:
hello I already try that but still got this

C:\Users\IamLexxx\Documents\Arduino\libraries\SPFD5408_TFT_Library\src\TftSpfd5408.cpp: In member function 'void TftSpfd5408::init()':
C:\Users\IamLexxx\Documents\Arduino\libraries\SPFD5408_TFT_Library\src\TftSpfd5408.cpp:115:3: error: 'textsize' was not declared in this scope
   textsize  = 1;
   ^~~~~~~~
C:\Users\IamLexxx\Documents\Arduino\libraries\SPFD5408_TFT_Library\src\TftSpfd5408.cpp:115:3: note: suggested alternative: 'textsize_x'
   textsize  = 1;
   ^~~~~~~~
   textsize_x

exit status 1

Compilation error: exit status 1

Event if i try examples tftbmp.ino