adafruit/TFTLCD-Library #49: Constructor is setting a digital pin before setting OUTPUT mode.

midi-mma, 2020-08-04:
In the constructor 
`Adafruit_TFTLCD::Adafruit_TFTLCD(uint8_t cs, uint8_t cd, uint8_t wr, uint8_t rd, uint8_t reset)`

we see the following lines of code:

```
  if (reset) {
    digitalWrite(reset, HIGH);
    pinMode(reset, OUTPUT);
  }
```

Is it valid to be setting the line to HIGH before the pinMode is set?  On SAMD51 the default pin mode is INPUT_PULLUP.

ladyada, 2020-08-04:
oh yeah thats an old Arduinoism - if you can submit a tested PR that would be great

midi-mma, 2020-08-04:
Unfortunately, I am unable to submit a tested PR at this time, because I am trying to get the HX8357 display working on the Grand Central M4, and I'm working with the code in the SAMD51 branch. Unfortunately my wiring does not match the breakout board exactly. But I noticed this issue while looking through the code.  Sorry!

makermelissa, 2020-08-05:
I believe the reasoning for this is to avoid dropping the reset line low between the 2 calls and resetting the processor.